levylabpitt/LV-Data #25: branches

ciozi137, 2020-07-11:
@johnbrickmaier @mayabostock 
The branches are a hot mess (probably my fault). I will try to fix and be in touch

ciozi137, 2020-07-11:
Actually, the branch structure is ok - I think I'm just confused:
[image]

Maybe I need to merge aws and commands into develop, and then we should branch from develop into "commands", "aws", "pg" branches for new features:

This is my goal:

[image]

looping @phil-shenk  in here too

phil-shenk, 2020-07-13:
aws was just a temporary branch to help merge in the S3 stuff a while back, and can be deleted I think, unless someone has been using it. I've been branching off from 'development' into things like 'development-issue21' then merging back into development to make my changes

ciozi137, 2020-07-13:
cool thanks for confirming

ciozi137, 2020-08-25:
I think this is fixed...

closed 2020-08-25 by ciozi137