levylabpitt/LV-Data #27: Command: Z Label

ciozi137, 2020-08-25:
?

mayabostock, 2020-08-28:
I'm not sure if I know what you mean by Z label

ciozi137, 2020-08-28:
Intensity or color scale label. The label VI has options for left/right/top/bottom, so I think we should add one for "Z". I'm not sure offhand what Igor calls it. I'll find out and post here.

ciozi137, 2020-08-28:
Command is **ColorScale**

ColorScale [flags] [, keyword = value, ...] [axisLabelStr ]
The ColorScale operation adds a color scale (or "color legend") annotation a graph, Gizmo plot or page layout. For background information, see Color Scales.
...
```
Image Plot ColorScale Example
Make/O/N=(20,20) img=p*q; NewImage img			// Make and display an image
ColorScale						// create default color scale
ColorScale/C/N=text0 nticks=3,minor=1,"Altitude"	// first annotation is text0
ModifyImage img ctab= {*,*,Relief19,0}			// 19-color color table
ColorScale/C/N=text0 axisRange={100,300}		// Detail for 100-300 range
ColorScale/C/N=text0 colorBoxesFrame=1			// Frame the color boxes
ColorScale/C/N=text0 frameRGB=(65535,0,0)		// Red frame
```

closed 2022-04-01 by ciozi137