levylabpitt/LV-Data #93: Reconcile reading DAT and TDMS data

ciozi137, 2022-12-08:
DAT requires this construction:
[image]

[key-value] where value = variant = [DBL]

TDMS:
[image]

[key-value] where value = variant = dict = key-value2 where value2 = variant = data (always [DBL]?)

ciozi137, 2023-01-21:
Fix looks like this:
[image]

closed 2023-01-21 by ciozi137

jsebolt, 2025-01-07:
@ciozi137 update the data analysis template to reflect these changes

jsebolt, 2025-01-07:
[image]
[image]